elrepo/packages #175: aic7xxx not working in RHEL/CentOS 7.5

msz59, 2018-07-16:
aic7xxx module in elrepo/packages/aic7xxx/el7 is marked as "rebuilt for RHEL 7.5" by _pjperry_. I tried to build it on freshly installed CentOS 7.5 (3.10.0-862.el7.x86_64 kernel) machine, to no success.

1. 'make' fails on aic7xxx_osm.c:

`/tmp/aic7xxx-kmod/el7/aic7xxx_osm.c: In function ‘ahc_linux_run_command’:`
`/tmp/aic7xxx-kmod/el7/aic7xxx_osm.c:1460:2: error: implicit declaration of function ‘blk_rq_tagged’ [-Werror=implicit-function-declaration]`
`  if (!blk_rq_tagged(cmd->request)`
`  ^`
`cc1: some warnings being treated as errors`

2. If overrided with -Wno-error=implicit-function-declaration, module gets built with warning:

`WARNING: "blk_rq_tagged" [/tmp/aic7xxx-kmod/el7/aic7xxx.ko] undefined!`

and then fails to load:

`insmod: ERROR: could not insert module aic7xxx.ko: Unknown symbol in module`

pjperry, 2018-07-16:
I get the same error.

Apologies, my fault. I pushed the update before checking the build compiled successfully. So far I have been unable to fix the issue so at present we are unable to support the aic7xxx module on EL7.5

pjperry, 2018-07-16:
I've rebuilt the package having fixed the compile error. Please would you be able to test? The updated packages are in the elrepo testing repository (currently syncing to the mirrors and should show up shortly). Thanks for reporting

msz59, 2018-07-17:
Built, installed, worked on current kernel.
Thank you!

pjperry, 2018-07-17:
Thanks for the feedback. I'll close this issue and move packages to the main repository. Thanks again.

closed 2018-07-17 by pjperry