elrepo/packages #186: no ath5k package in repo

akostadinov, 2019-01-26:
Hi, I see that ath5k package was available in repo but now I can't find it. Is this intentional?

toracat, 2019-01-26:
It's been archived (dated 2016). Or do you need it rebuilt for newer kernels?

akostadinov, 2019-01-26:
Yes, I tried to install on RHEL7 latest to no avail.

akostadinov, 2019-01-26:
Important is not to have wireless but to have my card working. Strange it broke given it seems still supported in upstream kernel. Perhaps due to backporting. In any case I would not ask anybody to spend time on this. Hopefully in RHEL8 it should work again because of the newer kernel rebase.

btw USB not always stable due to cooling and bad antennae.

closed 2019-01-26 by akostadinov

pjperry, 2019-01-26:
Red Hat chose not to include or support the ath5k driver in their RHEL7 kernel. We initially backported the driver for RHEL, but with each successive backporting of newer wireless stacks to their kernel, Red Hat make it harder for us to maintain a backport of this module as Red Hat only backport the bits their kernel needs, not the complete wireless stack. Don't get me wrong - it is doable, but IMHO just not something I consider is worth the time given you can purchase supported replacement hardware for a few bucks.
Also, it is very unlikely Red Hat will include support for ath5k in future releases. The ath5k is not supported in the RHEL8 beta kernel and I don't expect it to be supported in the final release of RHEL8.

toracat, 2019-01-26:
You can try kernel-ml or kernel-lt. ath5k is enabled there.

toracat, 2019-01-26:
Yes, probably. But if/when RH updates the wireless stack, it will be broken again.

toracat, 2019-01-27:
I confirm that the ath5k module (in-kernel) builds fine on RHEL 8beta.

akostadinov, 2019-01-27:
If it can be built in RHEL8 while it works without causing maintenance burden, that would be great. Thank you.