elrepo/packages #237: kmod-pata missing?

Livius90, 2021-01-23:
I have a JMicron JMB363 SATA+PATA PCIe card and as i see i need to use pata_jmicron to able to use it in CentOS 8 Linux.

I found that there is a [kmod-pata package](http://elrepo.org/tiki/kmod-pata) in ELRepo but yum can not find it when i try to install it. Only kmod-pata_amd, kmod-pata_sis and kmod-pata_via are available in yum. How can i install the full mod-pata package or just pata_jmicron from ELRepo?

toracat, 2021-01-23:
I'll try to build a kmod for you.

toracat, 2021-01-24:
The following kmod package and DUD iso image have been released:

kmod-pata_jmicron-0.1.5-1.el8_3.elrepo.x86_64.rpm
dd-pata_jmicron-0.1.5-1.el8_3.elrepo.iso

Please test. Feedback welcome.

Livius90, 2021-01-24:
Thanks that quick build.

I could find the mentioned .rpm in [this repo mirror](http://elrepo.reloumirrors.net/elrepo/el8/x86_64/RPMS/). I installed it manually by yum and now my JMicron JMB363 SATA+PATA PCIe card works fine (SATA was fine before, too), many thanks. I can see my PATA hard-disk drive in lsblk and lsscsi command, too. I am testing now with dd. I am coping zeros from /dev/zero to my /dev/sdc PATA drive, and it is working well at stabil 40-50 MB/s speed. Before this dd test, i could mount my NTFS partition too, so i think this .rpm package is ready to use everyone.

When it will be available in your released repo, to able to install it without any localinstall by yum?

pjperry, 2021-01-24:
Glad it works for you.

The files should be available on most of our mirror sites by now for yum. Just takes a few hours for all the mirror sites to update and sync.

Livius90, 2021-01-24:
And dmesg show this:

    [   15.978863] pata_jmicron: loading out-of-tree module taints kernel.
    [   15.978886] pata_jmicron: module verification failed: signature and/or required key missing - tainting kernel
    [   15.979104] pata_jmicron 0000:01:00.1: enabling device (0000 -> 0001)
    [   15.979506] scsi host10: pata_jmicron
    [   15.979604] scsi host11: pata_jmicron
    [   15.979650] ata11: PATA max UDMA/100 cmd 0xe040 ctl 0xe030 bmdma 0xe000 irq 17
    [   15.979652] ata12: PATA max UDMA/100 cmd 0xe020 ctl 0xe010 bmdma 0xe008 irq 17
    [   16.183485] ata11.01: ATA-5: MAXTOR 4K040H2, A08.1500, max UDMA/100
    [   16.183488] ata11.01: 78198750 sectors, multi 0: LBA
    [   16.205945] scsi 10:0:1:0: Direct-Access     ATA      MAXTOR 4K040H2   1500 PQ: 0 ANSI: 5
    [   16.206122] sd 10:0:1:0: [sdc] 78198750 512-byte logical blocks: (40.0 GB/37.3 GiB)
    [   16.206131] sd 10:0:1:0: [sdc] Write Protect is off
    [   16.206132] sd 10:0:1:0: [sdc] Mode Sense: 00 3a 00 00
    [   16.206146] sd 10:0:1:0: [sdc] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
    [   16.206188] sd 10:0:1:0: Attached scsi generic sg3 type 0
    [   16.216880] sd 10:0:1:0: [sdc] Attached SCSI disk

toracat, 2021-01-24:
Thanks for the testing and reporting back with the findings. Glad to learn things worked on your hardware. We will close this issue now.

closed 2021-01-24 by toracat

Livius90, 2021-01-24:
Is there any plan to build more Pata driver module .rpm into your repo? Many modules (probably all pata_xxx) were removed by Red Hat in RHEL 8. There are some popular SATA/IDE controllers like sata_nv, sata_sil, pata_sil680, sata_mv, pata_marvell which is still active products in any webshop.

pjperry, 2021-01-24:
If someone specifically needs them, then we can certainly look to build them and make them available, but we haven't just rebuilt everything Red Hat has removed / disabled for the sake of it.
Please keep in mind, that whilst these drivers may build now and we may be able to package and distribute them, there is no guarantee we will be able to continue to do so if Red Hat makes substantial changes around the code base in the RHEL kernel in the future. IOW, I would not be relying on these deprecated controllers in el8 for critical storage usage. We will continue to support them as long as we are able on a best effort basis.