elrepo/packages #238: kmod-r8188eu no more working on 8.3 version

enigma131, 2021-04-30:
sudo dnf --enablerepo="elrepo-testing" install kmod-r8188eu
...
depmod: WARNING: /lib/modules/4.18.0-240.el8.x86_64/extra/r8188eu/r8188eu.ko needs unknown symbol iwe_stream_add_event
depmod: WARNING: /lib/modules/4.18.0-240.el8.x86_64/extra/r8188eu/r8188eu.ko needs unknown symbol lib80211_get_crypto_ops
depmod: WARNING: /lib/modules/4.18.0-240.el8.x86_64/extra/r8188eu/r8188eu.ko needs unknown symbol wireless_send_event
depmod: WARNING: /lib/modules/4.18.0-240.el8.x86_64/extra/r8188eu/r8188eu.ko needs unknown symbol iwe_stream_add_point

But installed. Removing dkms, pulled out the dongle, rebooting,
then : dmesg -w , Pull in the dongle I got:

[ 454.338365] usb 1-5: new high-speed USB device number 8 using xhci_hcd
[ 454.464961] usb 1-5: New USB device found, idVendor=0bda, idProduct=8179, bcdDevice= 0.00
[ 454.464963] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 454.464964] usb 1-5: Product: 802.11n NIC
[ 454.464965] usb 1-5: Manufacturer: Realtek
[ 454.464966] usb 1-5: SerialNumber: 00xxxxxxxx
[ 454.479010] r8188eu: loading out-of-tree module taints kernel.
[ 454.479156] r8188eu: module verification failed: signature and/or required key missing - tainting kernel
[ 454.479267] r8188eu: Unknown symbol iwe_stream_add_event (err 0)
[ 454.479442] r8188eu: Unknown symbol lib80211_get_crypto_ops (err 0)
[ 454.479476] r8188eu: Unknown symbol wireless_send_event (err 0)
[ 454.479529] r8188eu: Unknown symbol iwe_stream_add_point (err 0)

Solution for now: building a dkms from Lwfinger's Git, but must take branch v4.1.8_9499 that is corrected now.
See : https://bugs.almalinux.org/view.php?id=57

Can the elrepo package be fixed ?

pjperry, 2021-04-30:
I have rebuilt the package against the latest RHEL8.3 kernel for you to test. The package is uploading to the testing repository and should be available on mirror sites shortly. Thanks

enigma131, 2021-05-01:
Thanks for quick response. I will test it next week

enigma131, 2021-05-03:
I've done a retry sudo dnf --enablerepo="elrepo-testing" install kmod-r8188eu
and got still the same error:
depmod: WARNING: /lib/modules/4.18.0-240.el8.x86_64/extra/r8188eu/r8188eu.ko needs unknown symbol iwe_stream_add_event
depmod: WARNING: /lib/modules/4.18.0-240.el8.x86_64/extra/r8188eu/r8188eu.ko needs unknown symbol lib80211_get_crypto_ops
depmod: WARNING: /lib/modules/4.18.0-240.el8.x86_64/extra/r8188eu/r8188eu.ko needs unknown symbol wireless_send_event
depmod: WARNING: /lib/modules/4.18.0-240.el8.x86_64/extra/r8188eu/r8188eu.ko needs unknown symbol iwe_stream_add_point

Have you corrected theses points ?

pjperry, 2021-05-03:
Yes, I see them too. This is not something we can fix, needs fixing upstream. I've removed packages from the repo as they are clearly not working.

pjperry, 2021-05-03:
Packages broken, removed from repo. Issue closed

closed 2021-05-03 by pjperry

toracat, 2021-05-03:
Reference point:

https://bugs.almalinux.org/view.php?id=57#c140

The source code in the Lwfinger's branch v4.1.8_9499 has since been all fixed and it builds and works without issues, according to the submitter. He uses dkms.

Problem is that it looksas if there is kABI breakage with every kernel update. So the question is more like ... is providing a kmod worthwhile?

pjperry, 2021-05-03:
> 
> Problem is that it looks as if there is kABI breakage with every kernel update. So the question is more like ... is providing a kmod worthwhile?

That's long been my thinking - it's just not worth the time and effort trying to support these Realtek devices. It would be far more beneficial to simply spend $10-20 buying a device that is natively supported by Linux.

For reference, here is the device I bought which is natively supported by Linux out of the box:

https://www.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/uk/wireless_adapters_n150/ew-7811un/

enigma131, 2021-05-03:
My device is working too 'out of the box' for Fedora , Debian and Ubuntu based distros (haven't try under Arch)
Wat would you mean by upstream? Who is in charge of this ?

pjperry, 2021-05-03:
The driver is in the staging area of the Linux kernel. Until it moves out of staging and into the kernel proper, there is little chance of it Red Hat enabling it in RHEL. So that means it won't ever be supported in RHEL7 or RHEL8, and it's unlike to be in RHEL9, which is already in development.
Which driver are Fedora , Debian and Ubuntu based distros using out of the box - the staging kernel driver? If those distros support your device, I would use one of those. Or if you must use RHEL, I would buy a $20 device that is supported.
Believe me, I feel your frustration - I have spent far too much of my time battling Realtek wireless drivers over the last 10+ years because they are not interested in supporting Linux users.

enigma131, 2021-05-04:
Seams all 3 distros use same source code : 
Fedora 33:

lsmod | grep 80211
lib80211               16384  1 r8188eu

modinfo r8188eu
filename:       /lib/modules/5.4.114-300.fc33.x86_64/kernel/drivers/staging/rtl8188eu/r8188eu.ko.xz
firmware:       rtlwifi/rtl8188eufw.bin
version:        v4.1.4_6773.20130222
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     50783CEE685A3C1A8A6B927
alias:          usb:v0BDApFFEFd*dc*dsc*dp*ic*isc*ip*in*
.....
depends:        cfg80211,lib80211
staging:        Y
retpoline:      Y
intree:         Y
name:           r8188eu
vermagic:       5.4.114-300.fc33.x86_64 SMP mod_unload 
sig_id:         PKCS#7
signer:         Fedora kernel signing key
sig_key:        2C:01:15:5F:B7:F4:E9:77:4D:41:BE:B5:0F:98:C7:66:49:08:63:50
....

Ubuntu 20.04.2:

lsmod | grep 80211
lib80211               16384  1 r8188eu

modinfo r8188eu
filename:       /lib/modules/5.4.0-72-generic/kernel/drivers/staging/rtl8188eu/r8188eu.ko
firmware:       rtlwifi/rtl8188eufw.bin
version:        v4.1.4_6773.20130222
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     620543B7F6A9685E8F044A4
alias:          usb:v0BDApFFEFd*dc*dsc*dp*ic*isc*ip*in*
....
depends:        cfg80211,lib80211
staging:        Y
retpoline:      Y
intree:         Y
name:           r8188eu
vermagic:       5.4.0-72-generic SMP mod_unload modversions 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        78:25:C2:0A:4E:41:60:7B:4A:4E:19:4A:8F:0C:8A:EE:31:4D:54:BF
...

Debian 10 (buster):

lsmod | grep 80211
lib80211               16384  1 r8188eu

sudo modinfo r8188eu
filename:       /lib/modules/4.19.0-16-amd64/kernel/drivers/staging/rtl8188eu/r8188eu.ko
version:        v4.1.4_6773.20130222
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     332134211D777E67B045044
alias:          usb:v0BDApFFEFd*dc*dsc*dp*ic*isc*ip*in*
...
depends:        cfg80211,usbcore,lib80211
staging:        Y
retpoline:      Y
intree:         Y
name:           r8188eu
vermagic:       4.19.0-16-amd64 SMP mod_unload modversions 
sig_id:         PKCS#7
signer:         Debian Secure Boot CA
sig_key:        4B:6E:F5:AB:CA:66:98:25:17:8E:05:2C:84:66:7C:CB:C0:53:1F:8C

lwfinger's git for my working DKMS:
https://github.com/lwfinger/rtl8188eu/tree/v4.1.8_9499

toracat, 2021-05-05:
I have earlier built a kmod package from the source code in the Lwfinger's branch v4.1.8_9499. It built and installed successfully.

$ modinfo /data/rpmbuild/elrepo/rtl8188eu/BUILD/rtl8188eu-v4.1.8_9499/rtl8188eu.ko
filename:       /data/rpmbuild/elrepo/rtl8188eu/BUILD/rtl8188eu-v4.1.8_9499/rtl8188eu.ko
version:        v4.1.8_9499.20131104
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
rhelversion:    8.3
srcversion:     ACC7B1A5514F62261F7F237
alias:          usb:v2357p0111d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Cd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep4008d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3311d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3310d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p330Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0DF6p0076d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p8179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*
depends:        cfg80211
name:           rtl8188eu
vermagic:       4.18.0-240.22.1.el8_3.x86_64 SMP mod_unload modversions

However, the problem is that, when test-installed on an EL 8 system, it does not create weak-links to other kernels than the one it was built against (in this case kernel-4.18.0-240.22.1.el8_3). There is kABI breakage, which is typical of Realtek drivers.

In other words, the kmod method is apparently not suitable here. Because you have been successfully building from the same source using dkms, I believe that that is the only viable solution.

toracat, 2021-05-05:
I noticed all alias lines were messed up. :(  Let's try again:

`alias:          usb:v2357p0111d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Cd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep4008d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3311d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3310d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p330Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0DF6p0076d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p8179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*`

enigma131, 2021-05-05:
Yes I had posted modinfos and cut into them with dots ....
Here a complete one : 

modinfo r8188eu
filename:       /lib/modules/5.4.115-300.fc33.x86_64/kernel/drivers/staging/rtl8188eu/r8188eu.ko.xz
firmware:       rtlwifi/rtl8188eufw.bin
version:        v4.1.4_6773.20130222
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     50783CEE685A3C1A8A6B927
alias:          usb:v0BDApFFEFd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pB811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0DF6p0076d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2C4Ep0102d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0111d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Cd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p331Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3311d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3310d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p330Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p18F0d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p8179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep4008d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*
depends:        cfg80211,lib80211
staging:        Y
retpoline:      Y
intree:         Y
name:           r8188eu
vermagic:       5.4.115-300.fc33.x86_64 SMP mod_unload 
sig_id:         PKCS#7
signer:         Fedora kernel signing key
sig_key:        08:6D:F6:C2:89:F8:EF:3F:D9:D6:76:BA:3D:16:DD:DE:E0:36:37:7E
sig_hashalgo:   sha256
signature:      35:2E:9A:0C:31:4D:AA:B2:E7:9F:2D:C4:01:9D:96:B3:78:F3:18:96:
		B5:6D:0A:E0:65:D0:96:6D:A5:AE:64:16:C5:54:09:0F:73:B6:7B:55:
		F7:A2:39:EE:78:4C:E5:0D:A7:A4:1B:01:0B:41:19:49:ED:21:17:C3:
		7E:87:2E:C5:00:AD:18:15:D0:15:D5:FB:96:98:66:FC:FE:CE:8C:2A:
		9C:8B:F7:79:B3:DB:1A:8A:4B:91:E7:EF:3C:1A:8A:D0:6F:64:67:18:
		18:64:F7:C3:CA:9F:79:9C:16:F2:7E:BF:B7:8D:2F:DC:7E:C5:CA:1C:
		11:DE:FE:D9:05:8B:F9:2B:EF:36:2E:2B:B7:C4:94:A1:A9:02:D3:66:
		CF:56:34:A9:08:BF:65:C4:FC:4C:DA:11:ED:E1:4E:7F:EE:32:09:89:
		E1:5D:A9:A3:B7:6E:37:B2:12:05:82:E4:57:8A:E7:58:35:C7:F4:12:
		F0:9B:45:28:A7:B3:71:D0:0F:C8:69:57:54:FC:0E:CC:D8:BF:8D:E3:
		95:20:A2:E2:02:5A:2F:AA:33:01:26:29:51:AA:76:86:AE:B4:79:7D:
		BF:14:E2:D9:A8:20:F0:EB:EC:3D:64:65:B0:91:1C:F2:33:0A:86:43:
		3A:72:4D:16:32:07:A6:18:97:9D:8E:57:92:DC:F0:FE:06:CD:67:C3:
		3E:97:67:F9:91:50:D3:0E:F5:58:C8:95:C1:87:01:67:05:52:E1:C9:
		E6:1B:72:A7:93:94:DF:66:31:B1:9D:91:C6:E7:65:E6:58:2F:DF:05:
		27:C7:A9:46:76:F8:9A:90:70:F1:1A:86:C6:E1:7E:00:1E:80:CB:FB:
		7D:1C:D1:4B:20:6E:69:4E:B6:EF:38:A1:A3:B2:5F:E0:AE:28:0D:20:
		D3:9D:FB:DD:4D:00:97:FD:BF:17:14:FB:24:4F:D2:21:88:50:74:57:
		68:5E:18:C9:70:73:0B:7B:EE:DF:A9:E6:94:A2:57:64:87:0B:F4:E7:
		B7:6D:0B:40:04:1E:DC:EF:70:B0:CD:3C:35:07:2C:98:22:20:99:8B:
		26:2D:B0:48:2D:7E:10:DC:0B:2E:50:16:C9:C4:83:E2:AC:A3:F4:9B:
		E8:46:07:00:20:15:C6:10:1C:67:93:B2:24:83:51:45:BE:87:CE:10:
		B1:8E:FE:C9:B4:13:18:27:E8:26:61:AB:82:F5:FE:C0:7E:FB:81:25:
		79:41:E1:A2:AF:A7:C8:26:66:9B:0E:2B:B0:EA:B3:9A:68:2E:45:B4:
		4D:E0:EE:4C:EC:06:AF:72:38:69:88:C9:7C:33:DC:76:A8:12:A0:04:
		11:1D:AF:DD:FC:43:AB:4C:09:F7:97:50
parm:           rtw_ips_mode:The default IPS mode (int)
parm:           ifname:The default name to allocate for first interface (charp)
parm:           if2name:The default name to allocate for second interface (charp)
parm:           rtw_initmac:charp
parm:           rtw_channel_plan:int
parm:           rtw_channel:int
parm:           rtw_wmm_enable:int
parm:           rtw_vrtl_carrier_sense:int
parm:           rtw_vcs_type:int
parm:           rtw_ht_enable:int
parm:           rtw_cbw40_enable:int
parm:           rtw_ampdu_enable:int
parm:           rtw_rx_stbc:int
parm:           rtw_ampdu_amsdu:int
parm:           rtw_power_mgnt:int
parm:           rtw_smart_ps:int
parm:           rtw_wifi_spec:int
parm:           rtw_antdiv_cfg:int
parm:           rtw_antdiv_type:int
parm:           rtw_enusbss:int
parm:           rtw_hwpdn_mode:int
parm:           rtw_max_roaming_times:The max roaming times to try (uint)
parm:           rtw_fw_iol:FW IOL (int)
parm:           rtw_mc2u_disable:int
parm:           rtw_80211d:Enable 802.11d mechanism (int)
parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)
parm:           debug:Set debug level (1-9) (default 1) (int)
parm:           monitor_enable:Enable monitor interface (default: false) (bool)

Concerning DKMS and Redhat's systems (including Fedora), you have to put at end of dkms.conf : 

NO_WEAK_MODULES="yes"

.. in order to have previous kernel working too. The inline manual was updating after my bug report here:

https://forums.fedoraforum.org/showthread.php?324697-Dkms-creates-wrong-symlinks-for-previous-kernel