elrepo/packages #240: kernel-ml RHEL9 plans?

akrherz, 2021-12-14:
I find the El Repo `kernel-ml` package very useful, thanks for providing it.  Are there plans to bring this package to RHEL9?  Thanks

pjperry, 2021-12-14:
Yes, we hope to be able to offer kernel-ml for RHEL9, although there are no plans to start working on it before the RHEL9 GA release, so it may follow a few months after the initial RHEL9 release. If you are interested in testing during the development process (really simple stuff - does it boot, any errors / stuff not working for you type stuff) please let us know as even simplistic testing and feedback on real hardware is invaluable.

akrherz, 2021-12-14:
Thank you for the response @pjperry , I did get the kernel-ml SRPM and try to rebuild it on RHEL9 without luck so far :(  Sorry for the silly question, but where is the kernel-ml repo even at?  I don't see it here?!?!

pjperry, 2021-12-14:
Alan, who maintains the kernel packages for elrepo, does not use github so you will not find them maintained online here. The SRPMs are your definitive sources.

akrherz, 2022-04-01:
Sorry to be annoying, but the closer of this issue means that kernel-ml for RHEL9 is still planned or is not?

toracat, 2022-04-01:
@pjperry's comment, "Yes, we hope to be able to offer kernel-ml for RHEL9" still holds true.

pjperry, 2022-04-01:
Nothing has changed since my answer of 14th December.
Yes there are plans to offer kernel-ml for RHEL9, but work on that will not start until RHEL9 GA has been released. I would then expect an initial release maybe 1-3 months after RHEL9 release. The thinking here is that upon initial GA release, the kernel in RHEL9 is relatively new and up to date, so there is no immediate or pressing need for a kernel-ml replacement offering. The older the release kernel becomes, the greater the need for a newer offering. Hence, we see no pressing requirement to ship kernel-ml immediately upon RHEL9 GA release.

akrherz, 2022-04-02:
@pjperry , while "pressing requirement" is hard to apply to us free-loading community users :), the `kernel-ml` package has significant usage due to the drivers that redhat disables in their kernel and the modules that do not exist as elrepo kmods or modules un-buildable against redhat's kernel.  For most of my hardware, `kernel-ml` is the easiest option :)

pjperry, 2022-04-02:
@akrherz acknowledged, and I get that. Notwithstanding, us free-loading community users are served by Alan, who for the last 10 plus years has maintained the elrepo kernel packages simultaneously for 2-3 distro releases, for free and in his own time, and has never taken a day off. As I've said above, all I can tell you is I can confirm Alan does plan to release a kernel-ml package for RHEL9, and his intentions are to start work on that once RHEL9 GA has been released. Judging by his previous efforts, it would be reasonable to expect a release around 1-3 months after RHEL9 GA although we make no promises as to how long that process may take. Anything you or I say here is not going to change Alan's plans but I can assure you, knowing Alan, he will work tirelessly to make that release available as soon as he is able (and you are welcome to test with his development builds). If you have an absolute requirement to install RHEL9 on unsupported hardware in the first month of release, then I would be talking with Red Hat now about your business use case for getting that hardware natively supported in RHEL.

pjperry, 2022-07-28:
Update: packages are being developed and we will be happy for folks to help test as soon as we have something installable.
See also: https://elrepo.org/bugs/view.php?id=1194

pjperry, 2022-09-17:
Update:
 I think we are pretty close to having something that is ready for limited public testing. We have package sets that build and install, but still have a minimal configuration rather than the usual elrepo enriched config. Alan is working on the configuration now and we will try to get some packages out for testing as soon as we are able for those who are able to test.

pjperry, 2022-09-18:
After much work, we finally have available a kernel-ml package set for el9 ready for testing. Packages for x86_64 and aarch64 are available for manual download and installation:

https://elrepo.org/people/ajb/devel/kernel-ml/el9/

For the first round of testing we are primarily concerned with installation, dependency and boot issues. Do packages install cleanly, any unexpected dependency issues for the combinations you are trying to install, and errors or warnings during or after booting. Please tell us what worked for you and what didn't. We expect these packages to install cleanly and be fully functional, subject to the enabled feature set (see below).

This first set of packages are based off the RHEL 9.0 kernel configuration. Some of the kernel options you may normally expect to find in an elrepo kernel offering may not have been enabled yet. For the second phase of testing, we will be asking for you to check, test and report on any features you would like added that may not currently be in RHEL. Just a clear list of the config options you would like enabled will be preferred (please no patches or config files of your own). For example, amateur radio is always disabled in the RHEL configurations (el{5|6|7|8|9}). So a simple request for "CONFIG_HAMRADIO=y" is all you need to do. Having end-users making configuration requests is how they can help out in the development process.

Providing feedback:
Please either provide feedback by email to one of the elrepo mailing lists (requires a subscription) or to the elrepo bug-tracker at https://elrepo.org/bugs/view.php?id=1194
We would prefer a new bug request for features you would like enabled as this makes things easier for us to track. If you post to a mailing list or elsewhere, we will open a bug request on your behalf to track progress.
New package sets will be released as per the upstream release cycle at kernel.org (normally a couple times per week), so any changes should make it into the next package set within a few days.

pjperry, 2022-10-04:
Kernel-ml packages for both aarch64 and x86-64 have now been released:

https://lists.elrepo.org/pipermail/elrepo/2022-October/006221.html

Closing this issue as resolved