elrepo/packages #241: kmod-nvidia-470xx not yet available?

mlewissmith, 2022-02-17:
Latest nvidia-detect (nvidia-detect-510.47.03-1.el7.elrepo.x86_64)

  nvidia-detect -v
  Probing for supported NVIDIA devices...
  [10de:11b4] NVIDIA Corporation GK104GL [Quadro K4200]
  This device requires the legacy 470.xx NVIDIA driver kmod-nvidia-470xx

But kmod-nvidia-470xx is not (yet) available in elrepo.  The previous legacy driver kmod-nvidia-390xx works for this card so the issue isn't pressing

wolfy67, 2022-02-17:
There are 7 different releases of kmod-nvidia-470 as well as 2 releases of kmod-nvidia-510 ( actually you used the version of nvidia-detect belonging to -510 ) as well as 2 releases of kmod-nvidia-460 

kmod-nvidia-460.80-1.el7_9.elrepo.x86_64.rpm | 30.9 MiB | 2021-May-12 10:36
-- | -- | --
kmod-nvidia-460.84-1.el7_9.elrepo.x86_64.rpm | 30.9 MiB | 2021-Jun-04 10:43
kmod-nvidia-470.103.01-1.el7_9.elrepo.x86_64.rpm | 48.2 MiB | 2022-Feb-01 19:13
kmod-nvidia-470.57.02-1.el7_9.elrepo.x86_64.rpm | 32.0 MiB | 2021-Jul-20 12:33
kmod-nvidia-470.63.01-1.el7_9.elrepo.x86_64.rpm | 48.2 MiB | 2021-Aug-11 22:15
kmod-nvidia-470.74-1.el7_9.elrepo.x86_64.rpm | 48.2 MiB | 2021-Sep-21 11:08
kmod-nvidia-470.82.00-1.el7_9.elrepo.x86_64.rpm | 48.2 MiB | 2021-Oct-28 20:21
kmod-nvidia-470.86-1.el7_9.elrepo.x86_64.rpm | 48.2 MiB | 2021-Nov-11 23:04
kmod-nvidia-470.94-1.el7_9.elrepo.x86_64.rpm | 48.3 MiB | 2021-Dec-14 15:59
kmod-nvidia-510.47.03-1.el7_9.elrepo.x86_64.rpm | 55.7 MiB | 2022-Feb-03 23:41
kmod-nvidia-510.54-1.el7_9.elrepo.x86_64.rpm | 55.7 MiB | 2022-Feb-15 18:20
 

[kmod-nvidia-460.80-1.el7_9.elrepo.x86_64.rpm](http://ftp.ines.lug.ro/elrepo/elrepo/el7/x86_64/RPMS/kmod-nvidia-460.80-1.el7_9.elrepo.x86_64.rpm)
	30.9 MiB	2021-May-12 10:36
[kmod-nvidia-460.84-1.el7_9.elrepo.x86_64.rpm](http://ftp.ines.lug.ro/elrepo/elrepo/el7/x86_64/RPMS/kmod-nvidia-460.84-1.el7_9.elrepo.x86_64.rpm)
	30.9 MiB	2021-Jun-04 10:43
[kmod-nvidia-470.103.01-1.el7_9.elrepo.x86_64.rpm](http://ftp.ines.lug.ro/elrepo/elrepo/el7/x86_64/RPMS/kmod-nvidia-470.103.01-1.el7_9.elrepo.x86_64.rpm)
	48.2 MiB	2022-Feb-01 19:13
[kmod-nvidia-470.57.02-1.el7_9.elrepo.x86_64.rpm](http://ftp.ines.lug.ro/elrepo/elrepo/el7/x86_64/RPMS/kmod-nvidia-470.57.02-1.el7_9.elrepo.x86_64.rpm)
	32.0 MiB	2021-Jul-20 12:33
[kmod-nvidia-470.63.01-1.el7_9.elrepo.x86_64.rpm](http://ftp.ines.lug.ro/elrepo/elrepo/el7/x86_64/RPMS/kmod-nvidia-470.63.01-1.el7_9.elrepo.x86_64.rpm)
	48.2 MiB	2021-Aug-11 22:15
[kmod-nvidia-470.74-1.el7_9.elrepo.x86_64.rpm](http://ftp.ines.lug.ro/elrepo/elrepo/el7/x86_64/RPMS/kmod-nvidia-470.74-1.el7_9.elrepo.x86_64.rpm)
	48.2 MiB	2021-Sep-21 11:08
[kmod-nvidia-470.82.00-1.el7_9.elrepo.x86_64.rpm](http://ftp.ines.lug.ro/elrepo/elrepo/el7/x86_64/RPMS/kmod-nvidia-470.82.00-1.el7_9.elrepo.x86_64.rpm)
	48.2 MiB	2021-Oct-28 20:21
[kmod-nvidia-470.86-1.el7_9.elrepo.x86_64.rpm](http://ftp.ines.lug.ro/elrepo/elrepo/el7/x86_64/RPMS/kmod-nvidia-470.86-1.el7_9.elrepo.x86_64.rpm)
	48.2 MiB	2021-Nov-11 23:04
[kmod-nvidia-470.94-1.el7_9.elrepo.x86_64.rpm](http://ftp.ines.lug.ro/elrepo/elrepo/el7/x86_64/RPMS/kmod-nvidia-470.94-1.el7_9.elrepo.x86_64.rpm)
	48.3 MiB	2021-Dec-14 15:59
[kmod-nvidia-510.47.03-1.el7_9.elrepo.x86_64.rpm](http://ftp.ines.lug.ro/elrepo/elrepo/el7/x86_64/RPMS/kmod-nvidia-510.47.03-1.el7_9.elrepo.x86_64.rpm)
	55.7 MiB	2022-Feb-03 23:41
[kmod-nvidia-510.54-1.el7_9.elrepo.x86_64.rpm](http://ftp.ines.lug.ro/elrepo/elrepo/el7/x86_64/RPMS/kmod-nvidia-510.54-1.el7_9.elrepo.x86_64.rpm)[](http://ftp.ines.lug.ro/elrepo/elrepo/el7/x86_64/RPMS/kmod-r8125-9.003.05-1.el7_8.elrepo.x86_64.rpm)
	55.7 MiB	2022-Feb-15 18:20

pjperry, 2022-02-17:
Yes, I haven't forked off for a legacy kmod-nvidia-470xx driver package yet. You can just use the last kmod-nvidia-470 release and not update - our yum-plugin-nvidia should actually prevent updating on hardware that is no longer supported.
I'll try to get around to releasing a legacy package for 470xx as soon as I'm able.

pjperry, 2022-03-04:
Resolved with this commit:

https://github.com/elrepo/packages/commit/7ca29fb3d8eaeb1013e420f89803bfa95da87a89

kmod-nvidia-470xx-470.103.01-1.el7_9.elrepo.x86_64.rpm
nvidia-x11-drv-470xx-470.103.01-1.el7_9.elrepo.x86_64.rpm
nvidia-x11-drv-470xx-libs-470.103.01-1.el7_9.elrepo.i686.rpm
nvidia-x11-drv-470xx-libs-470.103.01-1.el7_9.elrepo.x86_64.rpm

Packages are syncing to mirror sites and should be available through yum shortly.

If you are migrating from the non-legacy packages or (nvidia-390xx), you will need to uninstall those first, then install the 470xx legacy packages and reboot. Tested locally and they are working fine for me.

closed 2022-03-04 by pjperry