elrepo/packages #242: Is Elrepo broken for centos 8.0

uzairali001, 2022-02-28:
After installing elrepo in fresh Centos 8.0 `dnf` gives this error 
```
ELRepo.org Community Enterprise Linux Repository - el8                                  796  B/s | 520  B     00:00
Error: Failed to download metadata for repo 'elrepo': repomd.xml parser error: Parse error at line: 2 (Document is empty
)
```
```
uname -r
4.18.0-365.el8.x86_64
```

wolfy67, 2022-02-28:
On 2/28/22 14:28, Uzair Ali wrote:
>
> After installing elrepo in fresh Centos 8.0 dnf gives this error

centos 8 is dead and burried as it was EOLd 2 months ago

> ELRepo.org Community Enterprise Linux Repository - el8                                  796  B/s | 520  B     00:00
> Error: Failed to download metadata for repo 'elrepo': repomd.xml parser error: Parse error at line: 2 (Document is empty
> )
> uname -r
> 4.18.0-365.el8.x86_64

ElRepo for EL8 works just fine.

[root@pc69 ~]# yum clean all --enablerepo=elrepo
38 files removed
[root@pc69 ~]# yum list --disablerepo=\* --enablerepo=elrepo|less

ELRepo.org Community Enterprise Linux Repositor 230 kB/s | 170 kB     00:00    
Installed Packages

wolfy67, 2022-02-28:
You use an ELRepo mirror which has issues. My test above came from a box running Rocky.
Switch to a different mirror and you should be fine.

uzairali001, 2022-02-28:
> Thanks, I was able to install it with this command `sudo dnf install https://dl.fedoraproject.org/pub/epel/epel-release-latest-8.noarch.rpm`

Above command is installing only `epel`, I'm still unable to install `elrepo`. You suggested to switch different mirror can you show me how. I want run this command `dnf install kmod-wireguard wireguard-tools`

wolfy67, 2022-02-28:
Probably the easiest approach is to edit /etc/yum.repos.d/elrepo.conf and comment the line
mirrorlist=http://mirrors.elrepo.org/mirrors-elrepo.el8

Otherwise you could identify the mirror you actually used and ban it via any method you prefer.

uzairali001, 2022-02-28:
There is no such a file exists `/etc/yum.repos.d/elrepo.conf` neither in Centos nor in Rocky Linux instead there is `/etc/yum.repos.d/elrepo/elrepo.repo`, after editing and commenting out `mirrorlist=http://mirrors.elrepo.org/mirrors-elrepo.el8` it's working fine and I'm able to run `dnf install kmod-wireguard wireguard-tools`.

Thanks

closed 2022-02-28 by uzairali001

pjperry, 2022-11-27:
Our mirrorlist server has been under sustained attack for sometime. As mentioned earlier, if connection to our mirrorlist server is proving unreliable, please comment out the mirrorlist line and hard code a local mirror of your choosing which should eliminate the issue for you.