elrepo/packages #243: Update kmod-r8168 to latest 8.049.02 on EL8

kleisauke, 2022-03-06:
Thanks for providing `kmod-r8168` for EL8! We've a similar buggy NIC as the one reported in https://elrepo.org/bugs/view.php?id=1083.
```bash
$ lspci -nn | grep -i net
22:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)
```

Our server is provided by Hetzner, which mentions in [this article](https://community.hetzner.com/tutorials/installing-the-r8168-driver) that the distro r8169 driver can be used for kernel version 4.17 or later. Unfortunally, we still have network connectivity issues with this driver, despite running on kernel version 4.18. Disabling TCP segmentation offloading, as mentioned [in this article](https://docs.hetzner.com/robot/dedicated-server/operating-systems/realtek-r8169-r8168-nic/), doesn't seem to work either.

Fortunately, the r8168 driver (provided by ELRepo) seems to just work fine, except that we have encountered 3 kernel panics in the past 8 months (see the salient part of `kdump` in the attachment below). These panics were all on version 8.048.03.

I noticed that there is a newer version (8.049.02) of this driver available, which hopefully could fix the kernel panics we observed. Is it possible to update `kmod-r8168` to this new version?

(This request is similar to https://elrepo.org/bugs/view.php?id=1126, but then for EL8)

Attachment: [vmcore-dmesg-salient.txt](https://github.com/elrepo/packages/files/8193179/vmcore-dmesg-salient.txt)

toracat, 2022-03-06:
Acknowledged.

pjperry, 2022-03-06:
Done:

https://github.com/elrepo/packages/commit/af48828571513174599f0fd340c3fe67e10747e1

Updated packages are syncing to the mirror sites for release and should be available shortly.

Thanks

kleisauke, 2022-03-06:
Great, thanks for the quick response! I'll try this out soon.

Just wondering, is it also necessary to synchronize these definitions from the EL7 package?
```diff
--- a/r8168-kmod/el8/ELRepo-Makefile-r8168
+++ b/r8168-kmod/el8/ELRepo-Makefile-r8168
@@ -32,9 +32,15 @@ else
 # Called from kernel build system -- just declare the module(s).
 
 EXTRA_CFLAGS	+= -DCONFIG_R8168_NAPI
+EXTRA_CFLAGS	+= -DCONFIG_R8168_VLAN
+EXTRA_CFLAGS	+= -DCONFIG_ASPM
+EXTRA_CFLAGS	+= -DENABLE_S5WOL
+EXTRA_CFLAGS	+= -DENABLE_EEE
+EXTRA_CFLAGS	+= -DENABLE_USE_FIRMWARE_FILE
 
 obj-m		:= r8168.o
 
 r8168-objs	:= r8168_n.o r8168_asf.o rtl_eeprom.o rtltool.o
+r8168-objs	+= r8168_firmware.o
 
 endif
```
(since these capabilities are enabled by default in the official `src/Makefile`)

And, perhaps `-DCONFIG_DYNAMIC_ASPM` should be added to that as well? It looks like they enabled "dynamic ASPM" in their official `Makefile` too.

pjperry, 2022-03-07:
Sorry, I missed this - let me revisit it again today :-)

kleisauke, 2022-03-08:
Thanks! It seems to work:
```bash
$ dmesg | grep r8168
[    1.202636] r8168: loading out-of-tree module taints kernel.
[    1.202983] r8168: module verification failed: signature and/or required key missing - tainting kernel
[    1.205307] r8168 Gigabit Ethernet driver 8.049.02-NAPI loaded
[    1.218435] r8168: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    1.220489] r8168  Copyright (C) 2021 Realtek NIC software team <nicfae@realtek.com>
[    1.272750] r8168 0000:22:00.0 enp34s0: renamed from eth0
[    7.792093] r8168: enp34s0: link up
```
I'll close.

closed 2022-03-08 by kleisauke

kleisauke, 2022-03-13:
We received another kernel panic on one of our production servers with version 8.049.02 of this driver (with the same `transmit queue 0 timed out` error message, see the attachment below). 

I further investigated why we could not use the r8169 driver on our test server that has the same NIC. It seems that I was able to use the r8169 driver without any issues after doing these things:

1. Disable SG/TSO.
  ```bash
  # Not needed on kernel version v5.7 or later, see:
  # https://github.com/torvalds/linux/commit/95099c569a9fdbe186a27447dfa8a5a0562d4b7f
  nmcli conn modify $(nmcli -t c show --active | cut -d ":" -f 2) ethtool.feature-sg off ethtool.feature-tso off
  nmcli con up $(nmcli -t c show --active | cut -d ":" -f 2)
  ```
2. Set the ASPM performance policy or disable it completely.
  ```bash
  # https://bugzilla.redhat.com/show_bug.cgi?id=1671958
  grubby --update-kernel=ALL --args="pcie_aspm.policy=performance"
  # or: grubby --update-kernel=ALL --args="pcie_aspm=off"
  grub2-mkconfig -o /boot/grub2/grub.cfg
  ```
3. Disable MSI-X on everything with the ID `0x10ec:0x8168`.
  ```bash
  #!/bin/sh

  # Disables MSI-X on everything with the ID 0x10ec:0x8168, this
  # should not be needed on kernel version v4.19 or later.
  # See:
  # https://askubuntu.com/a/1203861
  # https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1779817
  # https://github.com/torvalds/linux/commit/083874549fdfefa629dfa752785e20427dde1511

  for i in /sys/bus/pci/devices/*; do
      if [ $(cat $i/vendor) = "0x10ec" -a $(cat $i/device) = "0x8168" ]; then
          echo "Disabling MSI on $i"
          echo 0 >$i/msi_bus
      fi
  done

  rmmod r8169
  modprobe r8169
  ```

Especially disabling MSI interrupts (3) for the NIC fixes this issue (this is also done by the r8168 driver, verified with `lspci -s <DEVICE> -vv`). This same issue also happens on the `kernel-ml` (`5.16.14-1.el8.elrepo.x86_64`) and `kernel-lt` (`5.4.184-1.el8.elrepo.x86_64`) packages. If this is a driver bug, perhaps it's already fixed with these commits (i.e. commits that were made on `drivers/net/ethernet/realtek` after v5.16)?
<details>
  <summary>Details</summary>

https://github.com/torvalds/linux/commit/1bd32771884187eacda045046805ca493b38e083
https://github.com/torvalds/linux/commit/5ef11c56b2332613cacd5e7ac17cfb1a073b66ab
https://github.com/torvalds/linux/commit/4e9c91cf92ec3acee94e04425594aa3688a225e8
https://github.com/torvalds/linux/commit/c75a9ad43691de040bead75f1924928111571f9c
https://github.com/torvalds/linux/commit/6d872df3e3b91532b142de9044e5b4984017a55f
https://github.com/torvalds/linux/commit/4b66d2161b8125b6caa6971815e85631cf3cf36f
https://github.com/torvalds/linux/commit/7462494408cd3de8b0bc1e79670bf213288501d0
https://github.com/torvalds/linux/commit/364ef1f3785785404473e02c6a020dae9d44fc4c
https://github.com/torvalds/linux/commit/6c8a5cf97c3f9d4c1a322c095ca60acf691d0226
https://github.com/torvalds/linux/commit/2d6600c754f8e4e52c392c3b5aecc554145e2524
https://github.com/torvalds/linux/commit/4b5f82f6aaef3fa95cce52deb8510f55ddda6a71
</details>

Anyways, I'm out of ideas and don't know how to investigate this further. For now, disabling MSI interrupts on this NIC seems to fix this, but it's something I'd rather avoid.

Attachment: [vmcore-dmesg-salient.txt](https://github.com/elrepo/packages/files/8239376/vmcore-dmesg-salient.txt)

wolfy67, 2022-03-13:
I've been using Realtek based network cards since the 8129 (!!!) era. My conclusion over time was that most Ethernet Realtek chips that I dealt with were of .. hmmmm ... let's call them  "debatable" (for lack of a more polite word) quality and I would never use them in a server.  They are cheap and heavily used in low cost motherboards because they do almost nothing in hardware and most of their "intelligence" is implemented in the software driver.
NDAs prohibit me of talking about what's inside an Intel chipset but please trust me, they are from a different universe when compared with Realtek.

If I were in your shoes I'd invest in add-on cards with Broadcom or Intel chipset, even 2nd hand. Of course,, assuming that is possible in your environment. 

wolfy, former maintainer of R81xx driver disks for CentOS 5

On March 13, 2022 1:03:11 PM GMT+02:00, Kleis Auke Wolthuizen ***@***.***> wrote:
> We received another kernel panic on one of our production servers with version 8.049.02 of this driver (with the same `transmit queue 0 timed out` error message, see the attachment below).
>
> I [...]