elrepo/packages #244: nvidia-detect -h

ntm, 2022-03-30:
Dear elrepo devs,
"nvidia-detect -h" doesn't print anything about exit status 8 and the 470.xx driver.
This is due to 2 small typoes in a single line here:
https://github.com/elrepo/packages/blob/45989aecaa1bc9f44ada8f0ac9ef7ed509afc879/nvidia-detect/nvidia-detect.c#L161

7 -> 8
and missing final semi-colon

It doesn't seem worth a fork + PR, but I can do that if you prefer. Thanks for the great resource!
Nicolas

pjperry, 2022-03-30:
Thanks, I'll queue up a fix and rebuild it as soon as I have a few spare moments (it's due a version bump anyway)

pjperry, 2022-03-30:
Fixed here:
https://github.com/elrepo/packages/commit/4171e3c26de3bb3741d5b5154763eac2f2e8b0c6

A new release (510.60.02) is syncing with our mirrors and should be available shortly

closed 2022-03-30 by pjperry