elrepo/packages #245: Install problem with `kmod-mvsas.x86_64`

mStirner, 2022-05-21:
Hello folks, im fresh installed a CentOS 9 Stream and having some trouble while installing the `kmod-mvsas.x86_64` driver:

```js
 yum install -y kmod-mvsas.x86_64
Letzte Prüfung auf abgelaufene Metadaten: vor 0:04:19 am So 22 Mai 2022 00:17:58 CEST.
Fehler: 
 Problem: widersprüchliche Anforderungen
  - nichts stellt kernel(module_layout) = 0x5d330546 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(__pci_register_driver) = 0xbe6cf13c bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(pci_unregister_driver) = 0x36ff01c4 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(pci_set_master) = 0x9b27d31d bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(dma_set_mask) = 0xa7a5e1e0 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(dma_set_coherent_mask) = 0x95500b59 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(dma_alloc_attrs) = 0xf3ce671a bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(dma_free_attrs) = 0xe634b46c bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(kmalloc_caches) = 0x142c0676 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(kmem_cache_alloc_trace) = 0x75fb3044 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(scsi_add_host_with_dma) = 0x826e7b9a bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(scsi_host_alloc) = 0x9e450066 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(pci_read_config_dword) = 0x0a098d1b bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(pci_disable_device) = 0xc9967dbb bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(scsi_scan_host) = 0xdac740bf bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(ata_scsi_dma_need_drain) = 0x6790c579 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(scsi_host_put) = 0x5e0861d2 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(scsi_remove_host) = 0xd20a0156 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(pci_enable_device) = 0x69236b5b bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(pci_write_config_dword) = 0x33cccc94 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(pci_request_regions) = 0xf52c7fe3 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(pci_release_regions) = 0xd52d9880 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(current_task) = 0x893dd803 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(pci_read_config_word) = 0x36cace3b bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(dev_printk) = 0xccc3c043 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(dma_pool_create) = 0x6f9df4c5 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_release_transport) = 0x954e466b bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(dma_map_sg_attrs) = 0xba9db02a bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(dma_unmap_sg_attrs) = 0x39e964c7 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_remove_host) = 0xeae36f12 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(put_device) = 0x00db1d92 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_bios_param) = 0x3ffb7df5 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_change_queue_depth) = 0x1dc05669 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_domain_attach_transport) = 0x456d423e bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_drain_work) = 0x80896f4b bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_eh_device_reset_handler) = 0x1f68f567 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_eh_target_reset_handler) = 0x28c463b6 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_get_local_phy) = 0xedb18a8f bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_ioctl) = 0x073eb419 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_notify_phy_event) = 0x8e01abef bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_notify_port_event) = 0xc0f8eee1 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_phy_reset) = 0xf4985475 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_queuecommand) = 0x4848fedb bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_register_ha) = 0xaeabeba2 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_slave_alloc) = 0xb431df79 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_slave_configure) = 0xd08774dc bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_ssp_task_response) = 0x0943057b bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_target_alloc) = 0x269e238e bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_target_destroy) = 0x5d81aa92 bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_unregister_ha) = 0x5f5d206d bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_alloc_slow_task) = 0x89b9c46c bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
  - nichts stellt kernel(sas_free_task) = 0x1161350e bereit, das von kmod-mvsas-0.8.16-1.el9_0.elrepo.x86_64 benötigt wird
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
```

It says "nothing provides kernel...."

I followed the instrucitons on your website and have imported the gpg key & installed the package:
```sh
rpm --import https://www.elrepo.org/RPM-GPG-KEY-elrepo.org
yum install https://www.elrepo.org/elrepo-release-9.el9.elrepo.noarch.rpm
```

And then a simple "yum install":
```sh
yum update
yum install -y kmod-mvsas.x86_64
```

Is there something im doeing wrong, or missing some dependencies (compiler/dev tools/etc?)
I even have installed "Development Tools". (`yum groupinstall "Development Tools"`)

Any help/hin is appreciated.

---

Im really thank full for your works.
Otherwise i could not use my Hardware.

Thank you very much!

pjperry, 2022-05-21:
What kernel are you running? (uname -a).
Elrepo kmod packages do not support CentOS Stream as the CentOS Stream kernel is not kABI compatible with the RHEL kernel. The CentOS Stream kernel is essentially a moving target to what will become the kernel in RHEL9.1. The packages just released are for RHEL9.0

mStirner, 2022-05-21:
`uname -a` output:
`Linux homeserver.lan 5.14.0-86.el9.x86_64 #1 SMP PREEMPT_DYNAMIC Fri May 6 12:02:49 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

Good that is just started from scratch and dint installed any other stuff.
What else do you recommend, install Rocky Linux?

wolfy67, 2022-05-21:
> Hello folks, im fresh installed a CentOS 9 Stream and having some trouble while installing the `kmod-mvsas.x86_64` driver:
>
> ```js
>  yum install -y kmod-mvsas.x86_64
> Letzte Prüfung auf abgelaufene [...]
[...]
> 
> It says "nothing provides kernel...."
> 
> I followed the instrucitons on your website and have imported the gpg key & installed the package:
> 
> ```shell
> rpm --import https://www.elrepo.org/RPM-GPG-KEY-elrepo.org
> yum install https://www.elrepo.org/elrepo-release-9.el9.elrepo.noarch.rpm
> ```
> 
> And then a simple "yum install":
> 
> ```shell
> yum update
> yum install -y kmod-mvsas.x86_64

Everything was correct so far, except the fact that you are using CentOS Stream 9. The packages from ELRepo are compatible with RHEL 9. Any compatibility with CentOS Stream 9 is pure luck.

> ```
> 
> (Even tried with `yum --disablerepo=\* --enablerepo=elrepo install kmod-mvsas.x86_64`)
> 
> Is there something im doeing wrong, or missing some dependencies (compiler/dev tools/etc?) I even have installed "Development Tools". (`yum groupinstall "Development Tools"`)
None of those packages is needed

> 
> Any help/hin is appreciated.

Either downgrade the kernel and use the one from RHEL 9.0 or install RHEL 9.0 altogether. The kernel(s) from CentOS Stream is(/are) newer and the packages built by ElRepo are not compatible.

pjperry, 2022-05-21:
As Wolfy says, the CentOS Stream kernel you are running is not compatible with elrepo packages, which are built for RHEL9.
Closing this issue as not something we can fix - you will need to run a compatible disto (RHEL9, Alma or Rocky etc, but not CentOS Stream).
If you want to run CentOS Stream then you should use the kmods SIG that CentOS provides, not elrepo.

closed 2022-05-21 by pjperry

mStirner, 2022-05-21:
@wolfy67 @pjperry Thanks for pointing that out to me.

And thanks for the awesome work you provide.
Last but not least, have a nice day :)