elrepo/packages #246: Megaraid_sas does not load in RockyLinux 8.6

TheCrockett, 2022-07-17:
Talks about versions not matching. 

I"ll update with the screenshot tommorrow.

pjperry, 2022-07-17:
Unable to replicate an issue on RHEL9:

$ rpm -q kmod-megaraid_sas
kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64

$ uname -a
Linux Build64R9 5.14.0-70.17.1.el9_0.x86_64 #1 SMP PREEMPT Tue Jun 14 11:32:10 EDT 2022 x86_64 x86_64 x86_64 GNU/Linux

$ lsmod | grep mega
megaraid_sas          184320  0

# dmesg | grep mega
[544417.986859] megaraid_sas: loading out-of-tree module taints kernel.
[544417.986983] megaraid_sas: module verification failed: signature and/or required key missing - tainting kernel
[544417.988910] megasas: 07.719.03.00-rc1

Everything looks fine so far - I cannot test the actual module as I do not have the hardware to test.

tqhoang84, 2024-08-13:
The megaraid_sas driver has been rebuilt many times for EL 8.6 -> 8.10.
https://elrepo.org/linux/dud/el8/x86_64/

closed 2024-08-13 by tqhoang84