elrepo/packages #247: kmod-wireguard on RHEL 8.4

afarresg, 2022-07-18:
Hello,

Up until a few weeks ago, kmod-wireguard was supported on RHEL 8.4. Since the removal of kmod-wireguard-1.0.20210606-1.el8_4.elrepo, it's no longer possible to install kmod-wireguard on RHEL 8.4. It's my understanding that RHEL 8.4 is a LTS release. Could it be possible to bring it back again?

Thank you

pjperry, 2022-07-18:
Userland utils packages are not dependent upon any one kernel release so should work across any point release whereas a kmod package delivers a kernel module that is dependent upon the kernel ABI (kABI) of that individual kernel release. If the kABI remains stable (i.e, kernel symbols used by the driver do not change) then the kmod driver will also work across multiple kernel releases (and RHEL point releases).
However, in reality Red Hat backports many changes into the kernel of new RHEL point releases that break the kABI compatibility, thus most kmod packages need to be rebuilt against the newer kernel and are thus only compatible with that newer kernel.

pjperry, 2022-07-18:
Closing issue as packages are available in the archive

closed 2022-07-18 by pjperry

alexisfrjp, 2022-11-11:
But why does it try to update a package for 8.7 whereas my rhel compat is 8.6?
AlmaLinux released 8.7 but Rocky Linux is still at 8.6.

```
Error: 
 Problem: cannot install the best update candidate for package kmod-wireguard-6:1.0.20220627-2.el8_6.elrepo.x86_64
  - nothing provides kernel >= 4.18.0-425.3.1.el8 needed by kmod-wireguard-7:1.0.20220627-3.el8_7.elrepo.x86_64
```

pjperry, 2022-11-11:
> But why does it try to update a package for 8.7 whereas I'm on 8.6? AlmaLinux released 8.7 but Rocky Linux is still at 8.6.
> 
> ```
> Error: 
>  Problem: cannot install the best update candidate for package kmod-wireguard-6:1.0.20220627-2.el8_6.elrepo.x86_64
>   - nothing provides kernel >= 4.18.0-425.3.1.el8 needed by kmod-wireguard-7:1.0.20220627-3.el8_7.elrepo.x86_64
> ```

What else would you propose? Should we leave RHEL 8.7 and Alma 8.7 users high and dry with a broken package because Rocky haven't released their clone yet?

It's up to you to manage your software - you can exclude kmod-wireguard until the required Rocky 8.7 kernel is available. DNF has given you the appropriate information as to the issue, it's up to you to manage that. Have you opened an issue with Rocky asking why they haven't released kernel-4.18.0-425.3.1.el8 yet?

alexisfrjp, 2022-11-11:
We can imagine it doesn't try to install a **el8_7** package (kmod-wireguard-7:1.0.20220627-3.**el8_7**.elrepo.x86_64) when the actual version of rhel is 8.6 only. Not like I'm trying to find a better way of handling things, just saying.

We know Rocky is always few days late, no need to open an issue.