elrepo/packages #248: kmod-tg3 for RHEL9 needed

quanmqiu, 2022-07-30:
Did you have any plans to release a kmod-tg3 package for RHEL 9？Thanks.

pjperry, 2022-07-30:
No, it's already natively supported in the distro kernel.
Do you have an issue with the in kernel driver? If so, have you filed a bug with Red Hat?

quanmqiu, 2022-08-16:
My R730 has 4 Broadcom 5720 NICs.  There is tg3 mod in the RHEL 9.0 installer iso. But no tg3 driver in the formal kernel releases.

kernel-core-5.14.0-70.13.1.el9_0.x86_64
kernel-core-5.14.0-70.17.1.el9_0.x86_64
kernel-core-5.14.0-70.22.1.el9_0.x86_64

All of above kernels can boot the machine, but no 5720 NICs recognition.

I unpacked those kernel packages, trying to search tg3 mod, and found nothing.

toracat, 2022-08-16:
For example, in kernel 5.14.0-70.22.1.el9_0, the tg3 module is:

/lib/modules/5.14.0-70.22.1.el9_0.x86_64/kernel/drivers/net/ethernet/broadcom/tg3.ko.xz

You can find this by running "modinfo tg3".

toracat, 2022-08-16:
By the way this module comes from

kernel-modules-5.14.0-70.22.1.el9_0

not from the kernel-core package.

quanmqiu, 2022-08-16:
Thanks for your advice.

The minimal installer iso didn't install the kernel-modules package. There is no NIC driver, then no network. I can't run dnf to install the kernel-modules package. It looks like the chicken and egg problem.

I will run dnf localinstall by using a usb-disk.

Thank you all.

toracat, 2022-08-16:
Which distro is this? RHEL does not have a minimal.iso. Or did you mean boot.iso?

quanmqiu, 2022-08-17:
Sorry. I mean minimal installation with boot iso.

toracat, 2022-08-17:
Thanks for the clarification.

closed 2022-08-21 by quanmqiu