elrepo/packages #250: megaraid_sas

Wobak, 2022-10-18:
Hello,

I can't install the megaraid_sas driver on Centos Stream 9 : 
```
yum info kmod-megaraid_sas.x86_64
Last metadata expiration check: 0:03:24 ago on Tue 18 Oct 2022 01:43:47 PM CEST.
Available Packages
Name         : kmod-megaraid_sas
Version      : 07.719.03.00
Release      : 1.el9_0.elrepo
Architecture : x86_64
Size         : 102 k
Source       : kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.src.rpm
Repository   : elrepo
Summary      : megaraid_sas kernel module(s)
URL          : http://www.kernel.org/
License      : GPLv2
Description  : This package provides the megaraid_sas kernel module(s).
             : It is built to depend upon the specific ABI provided by a range of releases
             : of the same variant of the Linux kernel and not on any one specific build.

```

My version:
```
# cat /etc/os-release
NAME="CentOS Stream"
VERSION="9"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="9"
PLATFORM_ID="platform:el9"
PRETTY_NAME="CentOS Stream 9"
ANSI_COLOR="0;31"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:centos:centos:9"
HOME_URL="https://centos.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux 9"
REDHAT_SUPPORT_PRODUCT_VERSION="CentOS Stream"
```

The installer fails with:
```
# yum -y install kmod-megaraid_sas.x86_64
Last metadata expiration check: 0:04:33 ago on Tue 18 Oct 2022 01:43:47 PM CEST.
Error:
 Problem: conflicting requests
  - nothing provides kernel(module_layout) = 0x5d330546 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(__pci_register_driver) = 0xbe6cf13c needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_unregister_driver) = 0x36ff01c4 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(param_ops_int) = 0x67054efe needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_set_master) = 0x9b27d31d needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(dma_set_mask) = 0xa7a5e1e0 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(dma_set_coherent_mask) = 0x95500b59 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(kmalloc_caches) = 0x142c0676 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(kmem_cache_alloc_trace) = 0x75fb3044 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(dma_alloc_attrs) = 0xf3ce671a needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(dma_free_attrs) = 0xe634b46c needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(_dev_info) = 0x66a98ad4 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_read_config_dword) = 0x0a098d1b needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_disable_device) = 0xc9967dbb needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(scsi_add_host_with_dma) = 0x826e7b9a needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(scsi_host_alloc) = 0x9e450066 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(_dev_err) = 0x2ba375c3 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(scsi_remove_host) = 0xd20a0156 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(scsi_scan_host) = 0xdac740bf needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(current_task) = 0x893dd803 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_write_config_dword) = 0x33cccc94 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(scsi_host_put) = 0x5e0861d2 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
[... 25 lines cut ...]
  - nothing provides kernel(scsi_print_command) = 0xd6345573 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pcie_capability_read_word) = 0x60c49938 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_enable_device_mem) = 0xe4979d4a needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_find_capability) = 0x646ce63a needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(blk_queue_flag_set) = 0x3bdff80e needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(__scsi_iterate_devices) = 0x284226d1 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(scmd_printk) = 0xecd3321e needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_irq_get_affinity) = 0x7582bc0c needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(_dev_notice) = 0x0867a696 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(blk_mq_pci_map_queues) = 0xb7981f51 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(blk_mq_unique_tag) = 0xb36e015e needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_release_selected_regions) = 0xdf507be9 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_request_selected_regions) = 0xce795f89 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_select_bars) = 0x68d2f310 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(blk_queue_update_dma_alignment) = 0x43da2bed needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(blk_queue_virt_boundary) = 0xfe10e370 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(fasync_helper) = 0xd40c3df4 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(kill_fasync) = 0x29bba27a needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_cfg_access_lock) = 0xb4dfa72c needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(pci_cfg_access_unlock) = 0xe2690ee0 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(scsi_host_busy) = 0x30ddcaed needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(driver_create_file) = 0x2a4aca43 needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
  - nothing provides kernel(driver_remove_file) = 0x71186ebd needed by kmod-megaraid_sas-07.719.03.00-1.el9_0.elrepo.x86_64
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
```

Wobak, 2022-10-18:
Seems that after a yum -y update kernel & kernel-devel were updated and are providing those.

They should be listed as dependency maybe?

closed 2022-10-18 by Wobak

pjperry, 2022-10-18:
They are listed as dependencies - that is what "nothing provides kernel(...)" means. Yum/dnf is telling you it is unable to find a compatible kernel that provides the kernel symbols required to meet those dependencies. That is because elrepo kmod packages do not support CentOS Stream kernels, only RHEL kernels.