elrepo/packages #252: kmod-drbd84 should be rebuilt against kernel-core-4.18.0-448.el8.x86_64

bubble75, 2023-03-21:
Fresh installation of Centos Stream 8
kernel 4.18.0-425.10.1.el8_7.x86_64 is not longer available, and install of kmod-drbd84 pulls in old kernel-core-4.18.0-408.el8.x86_64 because it requires kABI kernel(put_devmap_managed_page) = 0xde9d724c, which is missing in the current kernel-core-4.18.0-448.el8.x86_64.
weak-modules does not make drbd.ko available for that old version for some reason too.
So, after the installation I see no weak links:
find /lib/modules -name drbd.ko
/lib/modules/4.18.0-425.10.1.el8_7.x86_64/extra/drbd84/drbd.ko`
and it is impossible to use drbd at all, because it is compatible only with kernel which does not exist anymore.

toracat, 2023-03-21:
We, ELRepo, do not support CentOS Stream kernels. 

You may want to ask the CentOS kmods Sig. They provide kmods for CentOS Streams.

bubble75, 2023-03-21:
Thank you very much for the prompt answer. Then, we'd go to Rocky.

closed 2023-03-21 by bubble75