elrepo/packages #253: Another WireGuard breakage in the recent CS8 and future RHEL8 kernel releases

darkranger-red, 2023-04-10:
Since I don't see much discussion about this. So I think it should be good to submit an issue beforehand:
```
DKMS make.log for wireguard-1.0.20220627 for kernel 4.18.0-483.el8.x86_64 (x86_64)
Thu Apr  6 16:52:52 CST 2023
make: Entering directory '/usr/src/kernels/4.18.0-483.el8.x86_64'
  CC [M]  /var/lib/dkms/wireguard/1.0.20220627/build/main.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20220627/build/noise.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20220627/build/device.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20220627/build/peer.o
/var/lib/dkms/wireguard/1.0.20220627/build/peer.c: In function 'wg_peer_create':
/var/lib/dkms/wireguard/1.0.20220627/build/peer.c:61:40: error: macro "netif_napi_add" passed 4 arguments, but takes just 3
                        NAPI_POLL_WEIGHT);
```

So obviously a quick fix is simply remove 'NAPI_POLL_WEIGHT' from netif_napi_add(). And if anyone wants to see the patched repository, here is:
https://github.com/darkranger-red/wireguard-linux-compat/commit/f345e37a15d1232b1d8a8f2f38fed4de3f648579

pjperry, 2023-04-10:
ELRepo only support RHEL, not CentOS Stream.
CentOS Stream kernels are supported by the CentOS kmods SIG, so that is the place this should be fixed, and would then flow downstream from there.
The Wireguard project have stopped supporting CentOS Stream directly as patches were not being accepted to the CentOS/RHEL kernels.
ELRepo will only fix the issue for RHEL if it has not already been fixed upstream by the time RHEL 8.8 is released (in CentOS Stream). Please feel free to attach any patches here from CentOS Stream SIG

pjperry, 2023-04-10:
Just seen your patch - thanks :+1:

darkranger-red, 2024-06-11:
CS8 EOL after 05/31/2024.

closed 2024-06-11 by darkranger-red