elrepo/packages #257: drbd90-kmod at 9.1.x

zarganum, 2024-11-13:
Dear team, two questions on `drbd90-kmod`: 
1) any reason to keep it on 9.1.x?
2) any reason for vetoing the aarch64?

Starting from 9.2.6, DRBD replication now supports [kernel-level TLS](https://linbit.com/blog/encrypted-replication-with-drbd/) which might be essential for some environments.
Oracle is maintaining the [ktls-utils](https://github.com/oracle/ktls-utils) which is just passed its experimental status.

I tried your spec unlocking the aarch64 (removed `ExclusiveArch`) and changing the `Version` to 9.2.11. 
My express tests show it builds and works fine with TLS on `5.14.0-427.40.1.el9_4.aarch64` kernel.

toracat, 2024-11-13:
1. The 9.2.x branch has been published in the elrepo-testing repository. Regarding the 9.1.x branch, there will be 9.1.23 and that is supposed to be the last version. I plan to move 9.2.x to the main repository once 9.1.x comes to an end.

2. We do not have proper hardware to build and maintain kmod packages for aarch64 at this moment.

zarganum, 2024-11-13:
@toracat, thanks for swift response

re: `aarch64`: just in case, Oracle Cloud Infrastructure [Free Tier](https://www.oracle.com/cloud/free/) provides 3000 always-free ARM CPU hours per month (which is 4-cores), plus 24Gb of RAM. You can split it by cores/gigs in any combinations.

closed 2025-06-05 by toracat