elrepo/packages #258: kmod-usbip kernel panic when loading module on RHEL 9.6

kellard, 2025-06-04:
Hi, thanks for providing the package which I have used in earlier RHEL-versions but when upgrading to RHEL 9.6 and using the latest package from elrepo.org (v. 0.0.9) results in kernel panic for me when loading the kernel module `vhci-hcd`.

**Install package and load module**

- sudo dnf install kmod-usbip
- modprobe vhci-hcd

**Result**
Machine reboots with "kernel panic", se attached image.

### Specs
OS: RHEL 9.6 (x86_64)
kernel: 5.14.0-570.19.1.el9_6.x86_64

[image]

kellard, 2025-06-06:
I can confirm that the package is working fine!
Thanks a lot for the fast reply and fix!

tqhoang84, 2025-06-06:
That's great to hear that our fix worked!  Closing this issue now.

closed 2025-06-06 by tqhoang84