elrepo/packages #259: usbip-core.ko is missing in kmod-usbip since RHEL9.6

chenxiangw, 2025-06-25:
Hi
The file usbip-core.ko cannot be found in the latest kmod-usbip-0.0-10.el9_6.elrepo.x86_64.rpm, but before RHEL9.6 all kmod-usbip versions include this file.
Should this file be included in kmod-usbip or be dropped?

Best regards

tqhoang84, 2025-06-25:
Red Hat made changes to the kernel configuration in RHEL 9.6.  They now build the usbip modules and package them into an RPM called "kernel-modules-internal".  The problem is they don't have that RPM available to install.

The kernel's System.map has entries for usbip and that prevents us from relying on it since the modules can't be found.  Our initial workaround didn't work as expected and we decided to merge the usbip-core code into individual kmods (usbip-host.ko, vhci-hcd.ko).  See issue #258 for details.

With that said, are you just asking what happened to it or do you actually need usbip-core.ko for something else?

chenxiangw, 2025-06-26:
Thanks for your answering in full detail,
I just ask for the background about this change.

closed 2025-06-26 by chenxiangw

tqhoang84, 2025-06-26:
👍